adafruit/Adafruit_CircuitPython_LC709203F #3: simpletest fails on Feather nRF52840.

kattni, 2020-10-26:
simpletest fails to run on Feather nRF52840. Throws the following exception:
```
        if crc8 != self._buf[5]:
            raise RuntimeError("CRC failure on reading word")
```
Same hardware worked successfully on Feather M4, so hardware is verified as functioning.

ladyada, 2020-11-18:
@caternuson wanna try it again and slow down the clock, see if that helps?

caternuson, 2020-11-18:
@ladyada for Feather (this issue) or on RPI (other issue #4)?

ladyada, 2020-11-18:
feather - pi is resolved

caternuson, 2020-11-18:
gotcha. yah, i'll set this up and check.

caternuson, 2020-11-18:
Tried all the way down to 10kHz with CP 5.3.1. No change.

**400kHz** (default for CP 5.3.1)
```python
Adafruit CircuitPython 5.3.1 on 2020-07-13; Adafruit Feather Bluefruit Sense with nRF52840
>>> import board
>>> from adafruit_lc709203f import LC709023F
>>> sensor = LC709023F(board.I2C()) 
>>> sensor.cell_voltage
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "adafruit_lc709203f.py", line 112, in cell_voltage
  File "adafruit_lc709203f.py", line 171, in _read_word
RuntimeError: CRC failure on reading word
>>>
```

**100kHz**
```python
Adafruit CircuitPython 5.3.1 on 2020-07-13; Adafruit Feather Bluefruit Sense with nRF52840
>>> import board
>>> import busio
>>> from adafruit_lc709203f import LC709023F
>>> i2c = busio.I2C(board.SCL, board.SDA, frequency=100000)
>>> sensor = LC709023F(i2c)
>>> sensor.cell_voltage
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "adafruit_lc709203f.py", line 112, in cell_voltage
  File "adafruit_lc709203f.py", line 171, in _read_word
RuntimeError: CRC failure on reading word
>>> 
```

**10kHz**
```python
Adafruit CircuitPython 5.3.1 on 2020-07-13; Adafruit Feather Bluefruit Sense with nRF52840
>>> import board
>>> import busio
>>> from adafruit_lc709203f import LC709023F
>>> i2c = busio.I2C(board.SCL, board.SDA, frequency=10000)
>>> sensor = LC709023F(i2c)
>>> sensor.cell_voltage
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "adafruit_lc709203f.py", line 112, in cell_voltage
  File "adafruit_lc709203f.py", line 171, in _read_word
RuntimeError: CRC failure on reading word
>>> 
```

ladyada, 2020-11-18:
ok thankx - what an odd chip!

caternuson, 2020-11-18:
want me to scope this or test further?

ladyada, 2020-11-18:
nah

anecdata, 2021-01-06:
Adafruit Feather M4 Express with samd51j19
6.1.0-beta.2 on 2020-12-03
adafruit-circuitpython-bundle-6.x-mpy-20210106

I'm getting this with `simpletest.py`, can get around it by just the right re-seating of the JST 2 PH connectors:
```
LC709023F simple test
Make sure LiPoly battery is plugged into the board!
Traceback (most recent call last):
  File "code.py", line 8, in <module>
  File "adafruit_lc709203f.py", line 102, in __init__
  File "adafruit_lc709203f.py", line 135, in power_mode
  File "adafruit_lc709203f.py", line 195, in _write_word
  File "adafruit_lc709203f.py", line 195, in _write_word
OSError: [Errno 5] Input/output error
```
But then:
```
  File "code.py", line 1798, in <module>
  File "adafruit_lc709203f.py", line 124, in ic_version
  File "adafruit_lc709203f.py", line 184, in _read_word
RuntimeError: CRC failure on reading word
```

jameseleach, 2021-01-25:
I'm seeing this issue with a CLUE.   It doesn't happen every time I query `cell_voltage` or `cell_percent` but I do not see a pattern as to when it happens.   It happens at both the default clock speed and at 10,000Hz.

```python
Adafruit CircuitPython 6.1.0 on 2021-01-21; Adafruit CLUE nRF52840 Express with nRF52840
>>> import board
>>> import busio
>>> from adafruit_lc709203f import LC709203F
>>> i2c = busio.I2C(board.SCL, board.SDA, frequency=100000)
>>> sensor = LC709203F(i2c)
>>> sensor.cell_voltage
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "adafruit_lc709203f.py", line 114, in cell_voltage
  File "adafruit_lc709203f.py", line 184, in _read_word
RuntimeError: CRC failure on reading word
>>> sensor.cell_voltage
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "adafruit_lc709203f.py", line 114, in cell_voltage
  File "adafruit_lc709203f.py", line 184, in _read_word
RuntimeError: CRC failure on reading word
>>> sensor.cell_voltage
4.163
>>> sensor.cell_voltage
4.163
>>> sensor.cell_percent
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "adafruit_lc709203f.py", line 119, in cell_percent
  File "adafruit_lc709203f.py", line 184, in _read_word
RuntimeError: CRC failure on reading word
>>> sensor.cell_percent
98.4
>>> sensor.cell_percent
98.4
>>> 
```

TG-Techie, 2021-02-06:
I tried altering the simple_test.py to remove the `sensor.cell_voltage` read and the `sensor.cell_percent` reads start working. 

While trying to find the cause, I cleared the buffer before every read and write. Whenever the communication failed the buffer wasn't written and stayed cleared. Seems like the chip just isn't responding. I think the error might come from clock stretching, but that is merely speculation.

The datasheet says: (page 7)
```
* When you do not read CRC−8, LSI data is not reliable. CRC−8−ATM ex: (5 bytes) 0x16, 0x09, 0x17, 0xC2,
 0x0E → 0x86
```
Are they saying that the chip will send an invalid crc when it can't respond? Or just warning us to check the crc?

I'm not quite sure where to go from here but I thought these findings might be useful.

jerryneedell, 2021-02-23:
here is a simple comparison from an nRF52840 (CLUE) and a metro_esp32s2 runnin gthe lc709203f_simpletest with the same version of cp and same library -- clearly something is amiss on the nrf52840....

same LC709203F and battery -- just moved from  one board to the other

nrf52840 CLUE
```
Adafruit CircuitPython 6.2.0-beta.2-102-g2262af0b5 on 2021-02-22; Adafruit CLUE nRF52840 Express with nRF52840
>>> import lc709203f_simpletest
LC709203F simple test
Make sure LiPoly battery is plugged into the board!
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "lc709203f_simpletest.py", line 14, in <module>
  File "adafruit_lc709203f.py", line 124, in ic_version
  File "adafruit_lc709203f.py", line 184, in _read_word
RuntimeError: CRC failure on reading word
>>> 
```

metro_esp32s2
```
Adafruit CircuitPython 6.2.0-beta.2-102-g2262af0b5 on 2021-02-22; Adafruit Metro ESP32S2 with ESP32S2
>>> import lc709203f_simpletest
LC709203F simple test
Make sure LiPoly battery is plugged into the board!
IC version: 0x2717
Battery: 4.196 Volts / 100.0 %
Battery: 4.196 Volts / 100.0 %
Battery: 4.193 Volts / 100.0 %
Battery: 4.193 Volts / 100.0 %
```

jerryneedell, 2021-02-23:
arrgh !! I hooked up my logic analyzer to compare the behavior of the nrf52840 and the esp32s2.
on the first captures on the nrf52840 I saw the same CRC failure but after reconnecting the metro esp32s2 to get a "clean" capture and switching back to the nrf52840, I am unable to get the nrf52840 to fail....with or without the logic analyzer connected

```
Adafruit CircuitPython 6.2.0-beta.2-102-g2262af0b5 on 2021-02-22; Adafruit Feather Bluefruit Sense with nRF52840
>>> import lc709203f_simpletest
LC709203F simple test
Make sure LiPoly battery is plugged into the board!
IC version: 0x2717
Battery: 4.207 Volts / 100.0 %
Battery: 4.207 Volts / 100.0 %
Battery: 4.207 Volts / 100.0 %
```

It's getting late -- I will retry tomorrow....

jerryneedell, 2021-02-23:
Here are some screen shots from logic analyzer captures of the same sequence on the ESP32s2 (works) and CLUE (fails)

[image]

[image]

jerryneedell, 2021-02-23:
I added some code to print the CRC and read buffer
```
Adafruit CircuitPython 6.2.0-beta.2-102-g2262af0b5 on 2021-02-22; Adafruit CLUE nRF52840 Express with nRF52840
>>>
>>> import lc709203f_simpletest
LC709203F simple test
Make sure LiPoly battery is plugged into the board!
crc  0x75
['0x16', '0x11', '0x17', '0x17', '0x27', '0x75', '0x75', '0x0', '0x0', '0x0']
IC version: 0x2717
crc  0xa2
['0x16', '0x9', '0x17', '0x17', '0x27', '0x75', '0x75', '0x0', '0x0', '0x0']
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "lc709203f_simpletest.py", line 17, in <module>
  File "adafruit_lc709203f.py", line 114, in cell_voltage
  File "adafruit_lc709203f.py", line 186, in _read_word
RuntimeError: CRC failure on reading word
>>> 
```

jerryneedell, 2021-02-23:
here is another case where it got to the third read, but the read buffer also appears to contain data from the previous read

```
>>> import lc709203f_simpletest
LC709203F simple test
Make sure LiPoly battery is plugged into the board!
crc  0x75
['0x16', '0x11', '0x17', '0x17', '0x27', '0x75', '0x75', '0x0', '0x0', '0x0']
IC version: 0x2717
crc  0x99
['0x16', '0x9', '0x17', '0xa', '0x10', '0x99', '0x99', '0x0', '0x0', '0x0']
crc  0xed
['0x16', '0xf', '0x17', '0xa', '0x10', '0x99', '0x99', '0x0', '0x0', '0x0']
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "lc709203f_simpletest.py", line 17, in <module>
  File "adafruit_lc709203f.py", line 119, in cell_percent
  File "adafruit_lc709203f.py", line 186, in _read_word
RuntimeError: CRC failure on reading word
```

riscyd, 2021-03-20:
I ran into the same problem. Any fix would be great

e-simpson, 2021-06-13:
I am also experiencing the same issue on:

- latest 6.3.0 CircuitPython & libraries
- adafruit clue board (nrf52840)
- LC709023F connected via STEMMA

as mentioned here: https://github.com/adafruit/circuitpython/issues/3795#issuecomment-859962377

mrdalgaard, 2021-08-29:
Also seeing it on 7.0.0b0 with nRF52840 feather express

tannewt, 2021-08-30:
Should this be moved to the core?

mrdalgaard, 2021-09-01:
I tested this a bit more, by trying to use bitbangio as a workaround.
Out of the box, this doesn't work either, spitting out all sorts of different error:

```
>>> bus = bitbangio.I2C(board.SCL, board.SDA)
>>> fuelguage = LC709203F(bus)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/lib/adafruit_lc709203f.py", line 105, in __init__
  File "/lib/adafruit_lc709203f.py", line 146, in power_mode
  File "/lib/adafruit_lc709203f.py", line 220, in _read_word
OSError: [Errno 5] Input/output error

>>> fuelguage = LC709203F(bus)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/lib/adafruit_lc709203f.py", line 103, in __init__
ValueError: No I2C device at address: b

>>> fuelguage = LC709203F(bus)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/lib/adafruit_lc709203f.py", line 105, in __init__
  File "/lib/adafruit_lc709203f.py", line 146, in power_mode
  File "/lib/adafruit_lc709203f.py", line 220, in _read_word
OSError: [Errno 5] Input/output error

>>> fuelguage = LC709203F(bus)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/lib/adafruit_lc709203f.py", line 105, in __init__
  File "/lib/adafruit_lc709203f.py", line 146, in power_mode
  File "/lib/adafruit_lc709203f.py", line 220, in _read_word
OSError: [Errno 19] Unsupported operation

```
Lowering the frequency, will make it partially work. It will sometimes fail, and sometimes return a value.
```

>>> bus = bitbangio.I2C(board.SCL, board.SDA, frequency=1000)
>>> fuelguage = LC709203F(bus)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/lib/adafruit_lc709203f.py", line 105, in __init__
  File "/lib/adafruit_lc709203f.py", line 146, in power_mode
  File "/lib/adafruit_lc709203f.py", line 224, in _read_word
RuntimeError: CRC failure on reading word
>>> fuelguage = LC709203F(bus)
>>> fuelguage.cell_voltage
4.17
>>> fuelguage.cell_voltage
4.17
>>> fuelguage.cell_percent
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/lib/adafruit_lc709203f.py", line 124, in cell_percent
  File "/lib/adafruit_lc709203f.py", line 224, in _read_word
RuntimeError: CRC failure on reading word
>>> fuelguage.cell_percent
68.8
>>> fuelguage.cell_percent
68.8
>>> fuelguage.cell_percent
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/lib/adafruit_lc709203f.py", line 124, in cell_percent
  File "/lib/adafruit_lc709203f.py", line 224, in _read_word
RuntimeError: CRC failure on reading word
>>> fuelguage.cell_percent
68.8
>>> fuelguage.cell_percent
68.8
>>> fuelguage.cell_percent
68.8
>>> fuelguage.cell_percent
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/lib/adafruit_lc709203f.py", line 124, in cell_percent
  File "/lib/adafruit_lc709203f.py", line 220, in _read_word
OSError: [Errno 19] Unsupported operation

```

It does not seem to become entirely stable regardless of how low a frequency you specify.
Since busio on the nRF is locked in specific frequencies, it is not possible to test that with such a low speed.