adafruit/Adafruit_CircuitPython_LC709203F #4: simpletest fails on RPI

caternuson, 2020-10-30:
Re this thread:
https://forums.adafruit.com/viewtopic.php?f=19&t=171239

recreated locally:
```
(blinka) pi@raspberrypi:~ $ i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- 0b -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --                         
(blinka) pi@raspberrypi:~ $ python3 lc709203f_simpletest.py 
LC709023F simple test
Make sure LiPoly battery is plugged into the board!
Traceback (most recent call last):
  File "lc709203f_simpletest.py", line 14, in <module>
    print("IC version:", hex(sensor.ic_version))
  File "/home/pi/blinka/lib/python3.7/site-packages/adafruit_lc709203f.py", line 122, in ic_version
    return self._read_word(LC709203F_CMD_ICVERSION)
  File "/home/pi/blinka/lib/python3.7/site-packages/adafruit_lc709203f.py", line 171, in _read_word
    raise RuntimeError("CRC failure on reading word")
RuntimeError: CRC failure on reading word
(blinka) pi@raspberrypi:~ $ 
```

#3 related?

ladyada, 2020-10-30:
yah i probably have to see whats up - some simple math thing (cause it works on a feather m4)

stevewitz, 2020-11-13:
Hi,  I was wondering if this is a fixable problem.  I am inclined to think it is  a "repeated start" problem in the rpi.

Would really like to use this board in a project.
Thanks!
Steve

ladyada, 2020-11-15:
i just tried this on a pi4 and could not replicate this. 
are you sure the battery is plugged in? whats the wiring like?

caternuson, 2020-11-16:
(weird. i responded with something like this yesterday. not sure where it went. reposting)

I think my original test was with a Pi Zero W. Must have at least had a battery connected to get the I2C scan to work.

Could this be a brown out issue? I most likely just grabbed a LiPo laying around for that test. It may have been near dead. I just now charged up a LiPo and set this back up on a Pi 4 and it all seems to work.

[image]

```
(blinka) pi@raspberrypi:~ $ i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- 0b -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --                         
(blinka) pi@raspberrypi:~ $ python3 lc709203f_simpletest.py 
LC709023F simple test
Make sure LiPoly battery is plugged into the board!
IC version: 0x2717
Battery: 4.074 Volts / 87.4 %
Battery: 4.074 Volts / 87.4 %
Battery: 4.074 Volts / 87.4 %
^CTraceback (most recent call last):
  File "lc709203f_simpletest.py", line 19, in <module>
    time.sleep(1)
KeyboardInterrupt
(blinka) pi@raspberrypi:~ $ 
```

ladyada, 2020-11-16:
it *definitely* needs a battery attached to work

caternuson, 2020-11-16:
Yep. For sure. I rechecked the general behavior. No battery = nothing in I2C scan (expected). So not sure what gets you to I2C scan works but Python script does not.

jdimpson, 2020-11-16:
I'm having the same failure with an RPi Zero W, and a fully charged 18650 battery, wired similarly to caternuson's picture. I was able to enumerate it via I2C but get the same CRC error as reported. I will repeat using an RPi 3 or maybe 4, but not until next week.

ladyada, 2020-11-16:
yeah pleaes try again, we could not replicate it :/

caternuson, 2020-11-16:
I set this up with the battery running a CLUE (running altimeter code) and then ran the lc709203f_simpletest.py on the Pi4 and let it run until the battery ran down. It failed in what I would call the expected way. I2C comms finally fail and scan does not work either.

(ran for hours, tons of output snipped, only showing end result)
```pythBattery: 2.495 Volts / 0.0 %
Battery: 2.490 Volts / 0.0 %
Battery: 2.491 Volts / 0.0 %
Battery: 2.491 Volts / 0.0 %
Battery: 2.490 Volts / 0.0 %
Traceback (most recent call last):
  File "lc709203f_simpletest.py", line 17, in <module>
    "Battery: %0.3f Volts / %0.1f %%" % (sensor.cell_voltage, sensor.cell_percent)
  File "/home/pi/blinka/lib/python3.7/site-packages/adafruit_lc709203f.py", line 114, in cell_voltage
    return self._read_word(LC709203F_CMD_CELLVOLTAGE) / 1000
  File "/home/pi/blinka/lib/python3.7/site-packages/adafruit_lc709203f.py", line 180, in _read_word
    self._buf, self._buf, out_start=1, out_end=2, in_start=3, in_end=7
  File "/home/pi/blinka/lib/python3.7/site-packages/adafruit_bus_device/i2c_device.py", line 149, in write_then_readinto
    in_end=in_end,
  File "/home/pi/blinka/lib/python3.7/site-packages/busio.py", line 136, in writeto_then_readfrom
    stop=stop,
  File "/home/pi/blinka/lib/python3.7/site-packages/adafruit_blinka/microcontroller/generic_linux/i2c.py", line 88, in writeto_then_readfrom
    address, buffer_out[out_start:out_end], in_end - in_start
  File "/home/pi/blinka/lib/python3.7/site-packages/Adafruit_PureIO/smbus.py", line 273, in read_i2c_block_data
    ioctl(self._device.fileno(), I2C_RDWR, request)
OSError: [Errno 121] Remote I/O error
(blinka) pi@raspberrypi:~ $ i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --                         
(blinka) pi@raspberrypi:~ $
```

[image]

I'm going to move back to testing on a Pi Zero W in case there is something unique about that platform.

caternuson, 2020-11-17:
Interesting. Fails immediately on Pi 0 W. Battery charged for several hours. Restaged everything on Pi 0 W. And...

```
(blinka) pi@raspberrypi:~ $ i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- 0b -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --                         
(blinka) pi@raspberrypi:~ $ python3 lc709203f_simpletest.py 
LC709023F simple test
Make sure LiPoly battery is plugged into the board!
Traceback (most recent call last):
  File "lc709203f_simpletest.py", line 14, in <module>
    print("IC version:", hex(sensor.ic_version))
  File "/home/pi/blinka/lib/python3.7/site-packages/adafruit_lc709203f.py", line 122, in ic_version
    return self._read_word(LC709203F_CMD_ICVERSION)
  File "/home/pi/blinka/lib/python3.7/site-packages/adafruit_lc709203f.py", line 171, in _read_word
    raise RuntimeError("CRC failure on reading word")
RuntimeError: CRC failure on reading word
(blinka) pi@raspberrypi:~ $  
```

[image]

ladyada, 2020-11-17:
what if you catch that exception and try again immediately?

caternuson, 2020-11-17:
Were you thinking of trying that more low level, like in the lib somewhere?

Trying via a simple mod to the lc709203f_simpletest.py program:
```python
print("Reading IC Version...")
success = False
while not success:
    try:
        print("IC version:", hex(sensor.ic_version))
        success = True
    except:
        time.sleep(0.001)
print("DONE.")
```
And it just tries forever:
```
(blinka) pi@raspberrypi:~ $ i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- 0b -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --                         
(blinka) pi@raspberrypi:~ $ python3 lc709203f_simpletest.py 
LC709023F simple test
Make sure LiPoly battery is plugged into the board!
Reading IC Version...
```

ladyada, 2020-11-17:
hmm ok well at least its reproducable. ill find a pi zero today

ladyada, 2020-11-18:
try setting i2c speed lower on the pi zero like 50K or 10K - this chip does clock stretching which the older pi's dont like :(

[image]

stevewitz, 2020-11-18:
Setting the bus speed to 10000 works! (50000 doesn't).

Thanks for your help in getting this to work!!!  It's very much appreciated.

ladyada, 2020-11-18:
we'll add a note. the pi 4 handles clock stretching automagically btw

ladyada, 2020-11-18:
note added closing as resolved

closed 2020-11-18 by ladyada

caternuson, 2020-11-18:
Looks like this sensor has earned itself a special spot on The List:
https://learn.adafruit.com/i2c-addresses/troublesome-chips

ladyada, 2020-11-18:
ha!

jdimpson, 2020-11-25:
Thank you. I can confirm it works on Zero W, too. I followed the instructions here: [I2C Clock Stretching](https://learn.adafruit.com/circuitpython-on-raspberrypi-linux/i2c-clock-stretching) , setting i2c_arm_baudrate to 10KHz.

I assume changing the baud rate is necessary on all Pi other than RPi4 (and maybe 400). Is that right?

ladyada, 2020-11-25:
yes