adafruit/Adafruit_CircuitPython_LC709203F #11: Always get CRC error using bitbangio

wolfmanjm, 2021-03-16:
I am running on rpi zero w using Adafruit-Blinka 6.4.0 and adafruit-circuitpython-lc709203f 2.0.2 and adafruit-circuitpython-bitbangio 1.2.4 on raspbian buster python 3.7.

I am trying to use bitbangio as I also have an SSD1306 on the normal H/W I2C ports (and it won't work with the clock stretching needed with the lc7).

If I run the lc7 on the H/W I2C ports with clock stretching set to 10KHz it mostly works (I do get occasional CRC errors). However when trying to use the bitbangio I always get the CRC error, no matter what frequency I set for it. (I tried everything from 10KHz down to 100Hz).

The only change to the simple test is using i2c from bitbangio. I have tried multiple GPIO pin combinations (D23,D24 and D20, D21).

```
LC709203F simple test
Make sure LiPoly battery is plugged into the board!
Traceback (most recent call last):
  File "test-batmon-bitbang.py", line 17, in <module>
    print("IC version:", hex(sensor.ic_version))
  File "/home/pi/.local/lib/python3.7/site-packages/adafruit_lc709203f.py", line 127, in ic_version
    return self._read_word(LC709203F_CMD_ICVERSION)
  File "/home/pi/.local/lib/python3.7/site-packages/adafruit_lc709203f.py", line 187, in _read_word
    raise RuntimeError("CRC failure on reading word")
RuntimeError: CRC failure on reading word
```
I have read that this CRC error is quite common on different platforms, and I wonder if there are any hints as to how to make this work.

wolfmanjm, 2021-03-16:
I think this maybe an issue with bitbangio on rpi, as a logic analyzer trace of the exchange looks completely wrong to me.
[image]

wolfmanjm, 2021-03-16:
never mind, I think I need a pullup on SDA pin, but also I think ,maybe it is too much to expect linux/rpizero to be able to bitbang I2C. I think thi shas nothign to do with this driver at any rate. Sorry for the noise.

closed 2021-03-16 by wolfmanjm

dhalbert, 2021-03-16:
Let us know if you get it working.

wolfmanjm, 2021-03-17:
The pull up made no difference, and looking closely at the logic trace I now suspect the bitbangio is missing sending a final clock after the data where the ack should come in. This confuses my logic analyser (it says missing ack even though there is one clearly visible), and probably confuses the chip too.
I'll study this and see if there is a bug in the bitbangio routines, and if so I'll file an issue on that library.

Thanks

wolfmanjm, 2021-03-18:
FYI this turned out to be multiple bugs in the bitbangio I2C library that you have to use on blinka. I fixed them as best I could here https://github.com/adafruit/Adafruit_CircuitPython_BitbangIO/pull/18
It now works some of the time with this chip at around a frequency of 200Hz.
However the chip will still get confused and return CRC errors, which it also does when running at 10KHz from the H/W I2C.
I am still looking into that, I am not sure if it is a bug in this library or another bug in the bitbangio library.
I have logic traces when this happens on the bitbang i2c, but I also need to capture it on the h/w i2c and see if the issue is the same.