adafruit/Adafruit_CircuitPython_LC709203F #12: How to invoke the INT pin, or is this automatically triggered ?

wildestpixel, 2021-04-21:
I see this described as follows : 

`INT - Interrupt signal out, you can set this up to pull low when the voltage or percentage drops below a threshold. Pulled up to VIN with a 10K resistor`

Does this automatically trigger when IC senses battery too low, or is this programatically triggered ?

ladyada, 2021-04-21:
hi this isnt a bug report, tech support is handled in the forums. thanks :)

caternuson, 2022-10-17:
Reopening and changing title to better reflect issue. This library is missing functions to set registers 0x13 and 0x14 which control the INT pint behavior.

Arduino lib implementation for ref:
https://github.com/adafruit/Adafruit_LC709203F/blob/78acd68fd6915cd8646ed89fb02cac9351586174/Adafruit_LC709203F.cpp#L172-L190

jrrickerson, 2023-04-24:
I'm starting to look at this during the PyCon US 2023 sprints.

PR #25 (2023-05-01): Allow configuration of the low voltage alarm
Enable read/write properties for the 0x13 (low voltage RSOC threshold) and 0x14 (low voltage value register)
Fixes #12

NOTE: Untested so far on hardware due to lack of availability.
closed 2023-05-01 by tekktrik