adafruit/Adafruit_CircuitPython_LC709203F #18: ERROR: No matching distribution found for adafruit-circuitpython-lc709203f==2.2.1

anthonyclarka2, 2022-03-06:
Attempted to install adafruit-circuitpython-lc709203f on MacOS Catalina 10.15.7 via Python 3.10 installed from Homebrew.

Ran `pip install adafruit-circuitpython-lc709203f==2.2.1` which was copied from this page: https://pypi.org/project/adafruit-circuitpython-lc709203f/2.2.1/

Received the following error:

```
Collecting adafruit-circuitpython-lc709203f==2.2.1
  Using cached adafruit-circuitpython-lc709203f-2.2.1.tar.gz (28 kB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Installing backend dependencies ... done
  Preparing metadata (pyproject.toml) ... done
  WARNING: Requested adafruit-circuitpython-lc709203f==2.2.1 from https://files.pythonhosted.org/packages/fa/1b/9c3aa63a5c6c6c13894769e0df8e28dd7340dd53291a4cc459400edfd089/adafruit-circuitpython-lc709203f-2.2.1.tar.gz#sha256=ef046c8ea3090fc220692120b18cbd20d51c99f97b8c40b629e6e21b5115a3b4, but installing version 0.0.0
Discarding https://files.pythonhosted.org/packages/fa/1b/9c3aa63a5c6c6c13894769e0df8e28dd7340dd53291a4cc459400edfd089/adafruit-circuitpython-lc709203f-2.2.1.tar.gz#sha256=ef046c8ea3090fc220692120b18cbd20d51c99f97b8c40b629e6e21b5115a3b4 (from https://pypi.org/simple/adafruit-circuitpython-lc709203f/): Requested adafruit-circuitpython-lc709203f==2.2.1 from https://files.pythonhosted.org/packages/fa/1b/9c3aa63a5c6c6c13894769e0df8e28dd7340dd53291a4cc459400edfd089/adafruit-circuitpython-lc709203f-2.2.1.tar.gz#sha256=ef046c8ea3090fc220692120b18cbd20d51c99f97b8c40b629e6e21b5115a3b4 has inconsistent version: filename has '2.2.1', but metadata has '0.0.0'
ERROR: Could not find a version that satisfies the requirement adafruit-circuitpython-lc709203f==2.2.1 (from versions: 1.0.0, 1.0.1, 2.0.0, 2.0.1, 2.0.2, 2.0.3, 2.1.0, 2.1.1, 2.2.0, 2.2.1)
ERROR: No matching distribution found for adafruit-circuitpython-lc709203f==2.2.1
```

```
python --version 
Python 3.10.2

pip --version
pip 22.0.3 from /usr/local/lib/python3.10/site-packages/pip (python 3.10)
```

FoamyGuy, 2022-03-06:
Does anything different happen if you don't specify the version specifically?
```
pip install adafruit-circuitpython-lc709203f
```
I am not sure what could be causing the issue you see. I copy/pasted your install command and it worked successfully to install the library for me, but I am in different environment Pop_OS! Linux with Python 3.8.10.

FoamyGuy, 2022-03-06:
This SO answer may be relevant, it appears to be discussing the same error message albeit with a different library: https://stackoverflow.com/a/68126941/507810

anthonyclarka2, 2022-03-06:
The StackOverflow answer correctly installed the module, although its version is marked as 0.0.0:

```
python -m pip install --upgrade --no-cache-dir --use-deprecated=legacy-resolver adafruit-circuitpython-lc709203f
Collecting adafruit-circuitpython-lc709203f
  Downloading adafruit-circuitpython-lc709203f-2.2.1.tar.gz (28 kB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Installing backend dependencies ... done
  Preparing metadata (pyproject.toml) ... done
Requirement already satisfied, skipping upgrade: Adafruit-Blinka in /usr/local/lib/python3.10/site-packages (from adafruit-circuitpython-lc709203f) (7.1.0)
Requirement already satisfied, skipping upgrade: adafruit-circuitpython-busdevice in /usr/local/lib/python3.10/site-packages (from adafruit-circuitpython-lc709203f) (5.1.5)
Requirement already satisfied, skipping upgrade: adafruit-circuitpython-register in /usr/local/lib/python3.10/site-packages (from adafruit-circuitpython-lc709203f) (1.9.8)
Requirement already satisfied, skipping upgrade: Adafruit-PureIO>=1.1.7 in /usr/local/lib/python3.10/site-packages (from Adafruit-Blinka->adafruit-circuitpython-lc709203f) (1.1.9)
Requirement already satisfied, skipping upgrade: Adafruit-PlatformDetect>=3.13.0 in /usr/local/lib/python3.10/site-packages (from Adafruit-Blinka->adafruit-circuitpython-lc709203f) (3.20.1)
Requirement already satisfied, skipping upgrade: pyftdi>=0.40.0 in /usr/local/lib/python3.10/site-packages (from Adafruit-Blinka->adafruit-circuitpython-lc709203f) (0.53.3)
Requirement already satisfied, skipping upgrade: pyusb!=1.2.0,>=1.0.0 in /usr/local/lib/python3.10/site-packages (from pyftdi>=0.40.0->Adafruit-Blinka->adafruit-circuitpython-lc709203f) (1.2.1)
Requirement already satisfied, skipping upgrade: pyserial>=3.0 in /usr/local/lib/python3.10/site-packages (from pyftdi>=0.40.0->Adafruit-Blinka->adafruit-circuitpython-lc709203f) (3.5)
Building wheels for collected packages: adafruit-circuitpython-lc709203f
  Building wheel for adafruit-circuitpython-lc709203f (pyproject.toml) ... done
  Created wheel for adafruit-circuitpython-lc709203f: filename=adafruit_circuitpython_lc709203f-0.0.0-py3-none-any.whl size=5923 sha256=5e0b591303f41a58bd7e3a049091d7fdddb256828d3e1c0f677a0433a6fdfb8f
  Stored in directory: /Users/aclark/.tmp/pip-ephem-wheel-cache-v205et9a/wheels/02/e5/f0/d4ae0798c3ff7d8e29338bc25e62f65a3d83c359c6c11e8a0b
Successfully built adafruit-circuitpython-lc709203f
Installing collected packages: adafruit-circuitpython-lc709203f
Successfully installed adafruit-circuitpython-lc709203f-0.0.0
```

That does get things working for me, thank you! The rest of this comment is regarding the non-working command:

When omitting the version number, the behaviour is the same but it tries to download multiple versions in descending order:

```
python -m pip install adafruit-circuitpython-lc709203f
Collecting adafruit-circuitpython-lc709203f
  Using cached adafruit-circuitpython-lc709203f-2.2.1.tar.gz (28 kB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Installing backend dependencies ... done
  Preparing metadata (pyproject.toml) ... done
Discarding https://files.pythonhosted.org/packages/fa/1b/9c3aa63a5c6c6c13894769e0df8e28dd7340dd53291a4cc459400edfd089/adafruit-circuitpython-lc709203f-2.2.1.tar.gz#sha256=ef046c8ea3090fc220692120b18cbd20d51c99f97b8c40b629e6e21b5115a3b4 (from https://pypi.org/simple/adafruit-circuitpython-lc709203f/): Requested adafruit-circuitpython-lc709203f from https://files.pythonhosted.org/packages/fa/1b/9c3aa63a5c6c6c13894769e0df8e28dd7340dd53291a4cc459400edfd089/adafruit-circuitpython-lc709203f-2.2.1.tar.gz#sha256=ef046c8ea3090fc220692120b18cbd20d51c99f97b8c40b629e6e21b5115a3b4 has inconsistent version: filename has '2.2.1', but metadata has '0.0.0'
  Using cached adafruit-circuitpython-lc709203f-2.2.0.tar.gz (28 kB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Installing backend dependencies ... done
  Preparing metadata (pyproject.toml) ... done
Discarding https://files.pythonhosted.org/packages/08/b8/ec1ce9c832d469a38d0818b7fccba3f0f32d7060a0bae4cab6243be2a7ab/adafruit-circuitpython-lc709203f-2.2.0.tar.gz#sha256=fc951f3f50b99777acf01e63e2af4e6500d57739f791cc37a25e03084d044227 (from https://pypi.org/simple/adafruit-circuitpython-lc709203f/): Requested adafruit-circuitpython-lc709203f from https://files.pythonhosted.org/packages/08/b8/ec1ce9c832d469a38d0818b7fccba3f0f32d7060a0bae4cab6243be2a7ab/adafruit-circuitpython-lc709203f-2.2.0.tar.gz#sha256=fc951f3f50b99777acf01e63e2af4e6500d57739f791cc37a25e03084d044227 has inconsistent version: filename has '2.2.0', but metadata has '0.0.0'
  Using cached adafruit-circuitpython-lc709203f-2.1.1.tar.gz (28 kB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Installing backend dependencies ... done
  Preparing metadata (pyproject.toml) ... done
Discarding https://files.pythonhosted.org/packages/fd/19/f8fdd851ffe2b29ed76781b312fc1ab78b12a8874ed86af5ed7f8e75e814/adafruit-circuitpython-lc709203f-2.1.1.tar.gz#sha256=4c8258b6980594c07fdabe4e7f6b759ce98ec2f2c30e045d20f358e04817cee2 (from https://pypi.org/simple/adafruit-circuitpython-lc709203f/): Requested adafruit-circuitpython-lc709203f from https://files.pythonhosted.org/packages/fd/19/f8fdd851ffe2b29ed76781b312fc1ab78b12a8874ed86af5ed7f8e75e814/adafruit-circuitpython-lc709203f-2.1.1.tar.gz#sha256=4c8258b [... cut ...] .0.0.tar.gz#sha256=d32e1fe6d51d8e8d2390ea07b4ad36807cdcc129375ca33ab7cb1a979b487d44 has inconsistent version: filename has '1.0.0', but metadata has '0.0.0'
ERROR: Could not find a version that satisfies the requirement adafruit-circuitpython-lc709203f (from versions: 1.0.0, 1.0.1, 2.0.0, 2.0.1, 2.0.2, 2.0.3, 2.1.0, 2.1.1, 2.2.0, 2.2.1)
ERROR: No matching distribution found for adafruit-circuitpython-lc709203f
```

Thank you very much for your help, I really appreciate the super speedy response!

FoamyGuy, 2022-03-06:
Try like this:
```
pip install adafruit-circuitpython-lc709203f==2.2.1 --no-cache-dir
```
In your pasted output it shows that it used a cached version instead of downloading from pip. Someone in our discord was able to successfully install it on a Mac with Python 3.10 so I think maybe there is just something weird with the cached version you have for some reason.

FoamyGuy, 2022-03-06:
Nice, glad you were able to get it installed.

closed 2022-03-06 by FoamyGuy