adafruit/Adafruit_CircuitPython_LC709203F #20: Measuring battery voltage on a adafruit feather esp32-s3 tft  fails in circuitpython

BeatArnet, 2022-10-01:
- IDE: Thonny (german)
- Device: adafruit esp32-s3 tft
- circuitpython from end of september 2022
- LipO WITH 2000mAh ist attached 
- Using the code simpletest.py
- Libraries installed (including the latest versions of adafruit_lc709203f.pmy, adafruit_bus_device, adafruit_register)

Result:

**LC709203F simple test
Make sure LiPoly battery is plugged into the board!
Zurückverfolgung (jüngste Aufforderung zuletzt):
  Datei "<stdin>", Zeile 12, in <module>
  Datei "/lib/adafruit_lc709203f.py", Zeile 132, in __init__
  Datei "/lib/adafruit_lc709203f.py", Zeile 136, in init_RSOC
  Datei "/lib/adafruit_lc709203f.py", Zeile 257, in _write_word
OSError: [Errno 116] ETIMEDOUT**

Measuring the distance with the VL53L4CX, air quality with the bme688 and luminance with the bh1750 on the same esp32 works fine.   Detaching the sensors does not help.

BeatArnet, 2022-10-02:
Meanwhile I have found an additional notice:
The library adafruit_lc709203f reports, that there is no sensor at 0xb but I2C-Scan tells:
I2C addresses found: ['**0xb**', '0x23', '0x29', '0x77']

dhalbert, 2022-10-02:
The ESP32-S3 cannot use an LC709203F, because of an issue with its I2C implementation. The problem is specific to the S3.

https://github.com/adafruit/circuitpython/issues/6311
https://github.com/espressif/esp-idf/issues/8894#issuecomment-1125564756

This has been a very hard issue to diagnose. There is a problem deep in the ESP-IDF.

BeatArnet, 2022-10-02:
@dhalbert Thanks for the answer. Therefore I have to find another solution with the ADC.

dhalbert, 2022-10-02:
It works fine on the ESP32-S2, if you are willing to buy another board.

vladak, 2022-11-10:
Unfortunately, this bug seems to be hitting ESP32 V2 (running CircuitPython 8.0.0-beta.4) as well: https://forums.adafruit.com/viewtopic.php?p=947702#p947702